reprappro/RepRapFirmware #71: Z axis hangs while printing

gonAlonso, 2015-01-07:
Hello
Every time I print an object, the Z axis motor hangs after 1st print layer. I mean, it starts printing at first_layer_height, when moving on to the second one, it tryes to move, I hear it, but it barely moves. The funny is, I can move it through the web interface buttons, or sending g-codes. I've forced it to the maxSpeed (G1 Z20 F5000) and it works seamlessly. It looks like it doesn't compute acceleration the way it should. I've been browsing the reppo, the web, forums, and no solution by now. 
I'm using version 0.78y-dc42, is there an updated version?
Thks

RRP-support, 2015-01-07:
This is an automated reply to your message "[RepRapFirmware] Z axis hangs while printing (#71)" sent to support@reprappro.com.

Dear gonAlonso <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

RRP-support, 2015-01-09:
You're using dc42's version of the firmware, not this ours, so I don't know if that is causing the problem. Try reverting to the standard firmware (0.78c), using the default config.g, and see if you can recreate the problem. I would expect that you have your Z axis max speed or acceleration set too high; we have reduced it, with the following lines in config.g:

M201 X500 Y500 Z15 E500 ; Accelerations (mm/s^2)
M203 X15000 Y15000 Z100 E3600 ; Maximum speeds (mm/min)
M566 X1200 Y1200 Z30 E1200 ; Minimum speeds mm/minute

Try these settings.

Ian
RepRapPro tech support

gonAlonso, 2015-01-11:
Thank you so much for your reply, Ian.
It happens to be the same problem with the standard 78c FW. I've tested previous versions too without success.
The curious thing is, it worked nice from two weeks, but now Z axis stopped. I've been thinking on the changes I've made, and I realized I've started using slic3r instead of cura. And I tested it now, there seems to bee an option on slic3r that mess with the firmware, and cura doesn't. Did you know about any previous case? Is it known for you? Is there something I should set/unset on slic3r?
Thank you!

RRP-support, 2015-01-11:
This is an automated reply to your message "Re: [RepRapFirmware] Z axis hangs while printing (#71)" sent to support@reprappro.com.

Dear gonAlonso <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

RRP-support, 2015-01-12:
Generally, we recommend using Slic3r. You can use Cura, but there are a few workarounds required - there is a community-created script that switches Cura from absolute to relative extrusion, and changes the Z axis so it is moved separately from X and Y, rather than all together. However, I haven't heard of such problems using Slic3r, and as this isn't normal, I don't think this is a firmware issue - it's either mechanical, or a problem with the gcode you're generating. Make sure you are using our Slic3r settings profiles - https://reprappro.com/documentation/ormerod-2/printing/#Slic3r_profiles . What happens when you try to print the test files from the SD-Image, eg snowman.g?

Ideally, please email support@reprappro.com for this issue, as I don't think it is related to firmware. This is not the best channel for receiving support; it's for reporting firmware bugs.

Ian
RepRapPro tech support

closed 2015-01-13 by gonAlonso