reprappro/RepRapFirmware #73: Extruder not extruding 1.04 FW update

jeffreyb1004, 2015-03-17:
Extruder drive cannot be controlled in ponterface, reverting to .78c for now

RRP-support, 2015-03-17:
This is an automated reply to your message "[RepRapFirmware] Extruder not extruding 1.04 FW update (#73)" sent to support@reprappro.com.

Dear jeffreyb1004 <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

RRP-support, 2015-03-18:
The tool number for the first hot end, set by config.g has changed, from T1 to T0. Most likely, you're sending T1 to enable the hot end, but it's actually T0 you need to send. The first extruder should be enabled by default, after reset/power on. If your gcode selects T1, it deselects the active T0 extruder. Update your gcode files and Slic3r settings so it uses T0, or change config.g so it defines the first hot end as T1.
I'm not sure if the extruder has been temperature-limited in this version of the firmware. You may need to be at extrusion temperatures to extruder. Otherwise, send M302 to allow cold extrusion.
We're investigating problems with the web interface. It appears the firmware has trouble serving the files, at times, but it's not consistent.
Ian
RepRapPro tech support

jeffreyb1004, 2015-03-19:
Sorry maybe I wasn't clear, the extrude button on pronterface is not working with this new firmware thus is not driving the filament into the bowden tube nor reversing it out. this is not limited to cold extrusions as even when up to temp ~200C it still wont manually extrude/reverse. Printing still works though. web interface seems to be consistently down i.e. I can get to the web interface, but cannot connect to the machine. It must be noted that I am bridging the connection from my laptop but since I am able to physically get to the interface, it shouldn't be an internet connection error, just a printer connection one.

RRP-support, 2015-03-19:
This is an automated reply to your message "Re: [RepRapFirmware] Extruder not extruding 1.04 FW update (#73)" sent to support@reprappro.com.

Dear jeffreyb1004 <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

RRP-support, 2015-03-19:
We've tested this, and don't have a problem extruding from Pronterface. What Operating System are you using, and what version of Pronterface? Is it our version of Pronterface from the 'software' repository? Have you configured it correctly, ie set the length (mm to extrude), and speed (mm/s) to more than zero? What about if you send a gcode command to move the extruder, like 'G1 E10 F100'? Send M83 beforehand, to make sure the extruder is relative movement mode.
For the web interface, you need to clear the browser cache. If you've run the old version of the web interface, it caches some data, which the new version doesn't understand, so stops. In Chrome, attempt a connection to your printer, then Select Tools > Developer tools (or press F12), open the drop down 'Local Storage', select the IP address of your printer (which should show). You'll see two 'Keys'; 'settings' and 'temps'. Delete 'temps', or both of them. Shut the Chrome window, then try reconnecting. In Firefox, clear the settings by clearing Preferences > Advanced > Network, then clear Cached Web Content. In other browsers, you need to do something similar. Alternatively, changing the printer IP address will also work.
We're in the process of a heavy revision to the networking code, which will be much improved. Our firmware release schedule will be much more frequent than it has been, and issues like this will be corrected.
Generally, I don't think the issues you're experiencing are bugs, more issues with your setup. This 'Issues' part of github is for firmware bugs and feature requests, not for support. For further support, please email support@reprappro.com
Ian
RepRapPro tech support