reprappro/RepRapFirmware #74: 1.04 no longer possible to heat up extruder

mansj, 2015-03-21:
I'm using Simplify3D, and after firmware update today I can no longer turn on extruder one, and no temperatures are returned to the software. I can turn on the heat pad, but the temp is not reported back.

RRP-support, 2015-03-21:
This is an automated reply to your message "[RepRapFirmware] 1.04 no longer possible to heat up extruder (#74)" sent to support@reprappro.com.

Dear mansj <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

dc42, 2015-03-21:
RepRapPro renumbered the tools in version 1.04 so that the first extruder is numbered T0, instead of T1 as it as in version 0.78c. So, if you have taken the new config.g file from RRP's 1.04 release then the first extruder will now be T0; and if you have Cura configured to use T1 as the first tool, then the tool numbers in the gcode files it produces will be wrong. If this is the case, you can revert to your old config,.g file, or you can reconfigure Cura and generate new gcode.

You might also wish to look at my 1.04a-dc42 release, see http://forums.reprap.org/read.php?340,486462.

RRP-support, 2015-03-21:
This is an automated reply to your message "Re: [RepRapFirmware] 1.04 no longer possible to heat up extruder
 (#74)" sent to support@reprappro.com.

Dear dc42 <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

RRP-support, 2015-03-23:
In 1.04, the first hot end is assigned T0, rather than T1. Not an issue for new users, as all our settings and gcode that is supplied have also been updated. Users upgrading from older versions do need to be aware of this, of course, but can still define the hot end as T1 if they want, by editing config.g. Or, if in a single colour machine, define both T0 and T1 to use the same heater and extruder. Using T0 as the first tool has become the industry standard, and is what Repetier and Cura expect out of the box.

dc42 - I appreciate the help, but please don't promote your version of the firmware as an alternative to ours, in this 'Issues' section. That just muddies the waters. Particularly as your firmware, while it currently specifies T1, is going to change to T0, too. Also you've put a hard link into your reply, and if someone with this issue follows it, subsequent releases by you may be overlooked. Also, if the mansj replies to this thread, it may be an issue regarding your version, not ours!

Ian
RepRapPro tech support

mansj, 2015-03-23:
Thank you for the reply. I've got the extruder numbering down - however, the software I'm using (Simplify3D) seems to no longer be able to activate either extruder or monitor any temperature. 

I am aware that you can not support Simplify 3D, and indeed my Ormerod 2 is not on the list of supported hardware for Simplify 3D, but it WAS working fine until the new firmware.

RRP-support, 2015-03-23:
This is an automated reply to your message "Re: [RepRapFirmware] 1.04 no longer possible to heat up extruder
 (#74)" sent to support@reprappro.com.

Dear mansj <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

RRP-support, 2015-03-23:
Hi mansj

I can't think of anything that would specifically stop it working, but I don't have Simplyfy 3D available to test. Check in Pronterface or the web interface that you are getting a correct temperature, and the tool is enabled (ie that it is not reporting a temperature fault). If you want to try setting the hot end back to T1, in your config.g change the last few lines from:
M563 P0 D0 H1 ; Define tool 0
G10 P0 S-273 R-273 ; Set tool 0 operating and standby temperatures

to:
M563 P1 D0 H1 ; Define tool 1
G10 P1 S-273 R-273 ; Set tool 1 operating and standby temperatures

Let me know if that helps.
Ian
RepRapPro tech support