reprappro/RepRapFirmware #77: No "OK" reponse from a M105 request stops Octoprint connecting

OricTosh, 2015-03-29:
I'm running an Ormerod 2 with the latest firmware 1.04 and I wish to use octoprint running on a Raspberry Pi as my controller. Octoprint sends a M105 request upon starting a connection but doesn't get the "OK" so wont finish it's connection routine and show machine connected.

I posted about this on the octoprint mailing list 
https://groups.google.com/forum/?fromgroups#!topic/octoprint/vCrgYblhOWA
It would great if this was supported by rep-rap firmware

RRP-support, 2015-03-29:
This is an automated reply to your message "[RepRapFirmware] No "OK" reponse from a M105 request stops Octoprint
 connecting (#77)" sent to support@reprappro.com.

Dear Chris Hills <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

reprappro, 2015-03-31:
We've just checked and it does send ok.  Can you use a direct serial link (for example with Arduino Serial Monitor), send an M105 and paste the response you get?

Best wishes

RepRapPro Support Team

RRP-support, 2015-03-31:
This is an automated reply to your message "Re: [RepRapFirmware] No "OK" reponse from a M105 request stops
 Octoprint connecting (#77)" sent to support@reprappro.com.

Dear reprappro <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

OricTosh, 2015-04-01:
This was the response I got when I used a direct serial link
ok T:21.6 B: 21.8
Not sure what happening when this received by Octoprint, as it won't connect. Maybe it's the format and content of the string being sent or "what Octoprint is looking for" when it receives the data, that causes the problem. Will update the Octoprint mailing list with my M105 response.

Thanks
Chris

RRP-support, 2015-04-01:
This is an automated reply to your message "Re: [RepRapFirmware] No "OK" reponse from a M105 request stops
 Octoprint connecting (#77)" sent to support@reprappro.com.

Dear Chris Hills <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

OricTosh, 2015-04-01:
I'm now able to connect to my printer with OctoPrint and rep-rap firmware gives the correct response needed.

closed 2015-04-01 by OricTosh

tankboyben, 2015-07-09:
What was the solution

RRP-support, 2015-07-09:
This is an automated reply to your message "Re: [RepRapFirmware] No "OK" reponse from a M105 request stops
 Octoprint connecting (#77)" sent to support@reprappro.com.

Dear tankboyben <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.