reprappro/RepRapFirmware #78: web interface inactive

jasontrublu, 2015-04-25:
Hi,

I have an Omerod V1 and firmware from branch master at version 1.04 (104-02-02-2015).

the config file is read and the web interface is available. But is says "inactive" and I cannot do anything to connect and control the Omerod. What have I missed? (version 0.90 from 08.08.2014 was still working)

RRP-support, 2015-04-25:
This is an automated reply to your message "[RepRapFirmware] web interface inactive (#78)" sent to support@reprappro.com.

Dear Jason Peper <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

RRP-support, 2015-04-26:
Hi

There's a blog post with release notes for v1.04 here: https://reprappro.com/2015/03/16/reprap-firmware-version-1-04-released/

When updating from 0.78c (or other versions) to 1.04, there is an issue which needs the browser cache to be reset. There are instructions about clearing the correct part of the browser cache on that page. "Clear the cache in your browser. It may be caching javascript code or settings of the wrong version, or corrupt. For Chrome, clear the cache, particularly any downloaded files/site data: https://support.google.com/chrome/answer/95582?hl=en-GB . In Firefox, clear the settings by clearing Preferences > Advanced > Network, then clear Cached Web Content."

Alternatively, changing the ip address of the printer, in config.g, should also work.

Longer explanation: The web browser stores two strings relating to the printer's ip address, 'settings' and 'temps'. The 1.04 web interface has a different string for 'temps', so if there is a pre-existing version (from 0.78c), it doesn't recognise it, and fails to load. There isn't any error checking (there is in the next version of the firmware) so at the moment you have to delete the string manually (by deleting cookies/downloaded web content), so the web interface can create a new version.

If you want to delete this directly in Chrome, rather than delete all cookies etc, open Chrome > Tools > Developer tools, click 'Resources', click the arrow to reveal the drop down under 'Local Storage'. Then connect to the web interface. The ip address should show below 'Local Storage'. Click on this, and it should show the 'settings' and 'temps' variables. Delete the 'temps' variable, then refresh the connection to the web interface.

Finally, some users can't connect to the Duet and get lots of network errors in the console log (error code -11) in Pronterface. Reset your TCP/IP stack: https://support.microsoft.com/en-us/kb/299357

If you want to revert to older firmware, I have added a 'tag' and 'release' of v0.78c in the github repository, here: https://github.com/reprappro/RepRapFirmware/releases

I'm not sure of the status of firmware 0.90... it wasn't an official release. If you want to test our next firmware (v1.09) it's in the 'test' branch of the github repository, and is due for imminent release. This has much improved network code and web interface, courtesy of zombiepantslol.

I hope that helps.

Ian
RepRapPro tech support

jasontrublu, 2015-04-27:
thanks a lot! That worked and I'll subscribe to the release blog from now on!

RRP-support, 2015-04-27:
This is an automated reply to your message "Re: [RepRapFirmware] web interface inactive (#78)" sent to support@reprappro.com.

Dear Jason Peper <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

RRP-support, 2015-04-27:
Great, glad that helped. Firmware v1.09 will be released in the next couple of days (just adding the finishing touches), and is a big improvement on the networking, both reliability and speed. Keep an eye on the blog!

Ian
RepRapPro tech support

closed 2015-04-27 by RRP-support