reprappro/RepRapFirmware #79: Does not work with arduino

JelleB, 2015-05-08:
Somewhere along the line of development, the requirement to built the firmware with plain arduino got lost, I think that is a very bad thing.
The board is based on the Arduino due not only because those designs were available, but presumably also because Arduino is a very low threshold entry into programming. That makes it very easy for unskilled enthusiasts to compile and change Arduino programs themselves, but also to contribute those changes back to the community. 
I count 11 hurdles to take (some with substeps a) to e)) to built this firmware. In contrast, to compile Marlin, you only have to download the source and click on the .ino.  Now tell me, which firmware is easier to hack, and which one will get the most contributions? 
Don't get me wrong, it is very understandable why you'd rather go to eclipse if you are doing this fulltime. There is nothing wrong with that. What is wrong, is that in doing so, apparently support for compiling with Arduino got lost. For any OSS project to succeed, you need mindshare. Arduino is a very successful project, just because of that mindshare. Basing the duet on the due was a very good move. Moving away from arduino for the firmware negates all that.
So, please add a .ino that allows you to compile with arduino?

RRP-support, 2015-05-08:
This is an automated reply to your message "[RepRapFirmware] Does not work with arduino (#79)" sent to support@reprappro.com.

Dear Jelle Boomstra <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

RRP-support, 2015-05-11:
Hi JelleB
Thanks for the feedback. We are aware this is an issue, and are working to get the firmware compiling again under Arduino IDE. The main problem has been recent changes to the Arduino IDE; it was compiling, but now we need to make further changes to get it working again.
Ian
RepRapPro tech support

chrishamm, 2015-05-11:
The main reason RepRapFirmware does not compile with the stock Arduino IDE lies in its dependency to third-party libraries, which are still way easier to integrate in Eclipse. Please keep in mind that different libraries like LWIP and parts of the ASF, which are not included in the original Arduino package, are needed to get the Ethernet interface working and to utilise other features the rather new SAM8XE 32-bit processor. Marlin OTOH does not have such dependencies, however its functionality and power is very limited compared to RepRapFirmware, which also runs on an entirely differnet architecture (AVR vs. ARM).

Besides, since Arduino v1.5.8 the Arduino Due package had been dropped from the core package and they did not provide a nice way to integrate other Arduino-compatible boards from their Boards Manager. Fortunately they've added a new way to do so to their v1.6.4 release, so I will write a new package file for the Duet over the next few days and update the build instructions once again. I hope this will simplify the build process, and hopefully restore the ability to compile the firmware via the Arduino IDE. I can't guarantee I'll get the latter part working, though, but I'll definitely try.

RRP-support, 2015-05-11:
This is an automated reply to your message "Re: [RepRapFirmware] Does not work with arduino (#79)" sent to support@reprappro.com.

Dear Christian Hammacher <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

chrishamm, 2015-05-24:
Alright, meanwhile I can say that this would be mission impossible! The reason is that the networking stack (LWIP) has a too complex directory structure for Arduino. It cannot be built with Arduino IDE, although all other libraries can be built just nicely.

Thanks to the changes introduced by Arduino 1.6.4, we can now at least provide the Duet files as an individual boards package (see https://github.com/chrishamm/RepRapFirmware/tree/dev/Hardware). However, a build script is needed to compile the firmware due to the problem I mentioned above. I've updated 3D-ES's Bash script to take care of the required changes (see https://github.com/chrishamm/RepRapFirmware/blob/dev/3d-es-make.sh), however for the moment it cannot be used under Windows, but will work on Linux and most likely on OS X.

I will come up with a more generic approach (some kind of Makefile) in near future, but even then the user will have to install Make on a Windows environment (presuming it isn't already installed by VC++ or so). I wish there was a way to use Arduino IDE to build it entirely, however as I already explained this is (currently) not possible. Sorry!