reprappro/RepRapFirmware #80: Webinterface: layer statistics graph

hieronymousch, 2015-05-15:
Hi,

Very much loving the new webinterface (ditched pronterface & repetier). 
One small detail: webinterface doesn't update the layer statistics graph as you can see in the screenshot.
Working in FireFox.

[image]

RRP-support, 2015-05-15:
This is an automated reply to your message "[RepRapFirmware] Webinterface: layer statistics graph (#80)" sent to support@reprappro.com.

Dear hieronymousch <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

RRP-support, 2015-05-15:
Hi  hieronymousch
How are you generating gcode? Slic3r, Cura or some other gcode editor? Or through Repetier Host?
To get the layer statistics working, the web interface has to find the last Z move in the gcode file, which defines the object height. We've noticed a small bug in the web interface that causes it to ignore the last Z move if it is followed by a 'comment' (a text note preceded by ';'). This can happen if you have verbose Gcode turned on (Slic3r > Print Settings > Output options > Output file > Verbose G-code). This puts a 'comment' (text preceded with';') on every line, so the web interface cannot determine the object height.
Also, files which have been uploaded by USB (eg through Pronterface) have comments removed; the filament usage is held in a comment, so this gets deleted. The web interface will report just file usage and layer time statistics, then.
If you can post a portion of your gcode, or email it to me at support@reprappro.com , I'll take a look and see if it's anything else that may be causing it.
Ian
RepRapPro tech support

hieronymousch, 2015-05-15:
Hi,

Thanks for your quick reply. I indeed used slic3r (version 1.26) but not with the verbode G-code optiion. I do however get some comments on the end of the G-code file.
I mailed a small gcode file to support email.

Thanks,

J

RRP-support, 2015-05-15:
This is an automated reply to your message "Re: [RepRapFirmware] Webinterface: layer statistics graph (#80)" sent to support@reprappro.com.

Dear hieronymousch <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

RRP-support, 2015-05-27:
Hi hieronymousch
I have replied by email, but I think the problem is you are using 'Lift Z'; ie between each move, the Z moves up, moves in X and Y, then down again, and does the same thing at the end of each layer. This is set in Slic3r > Printer Settings > Extruder 1 > Retraction > Lift Z. We normally have this set to 0. I'd guess having this set messes up the layer statistics! I get the same thing as you when I print this gcode. I've sent it to our programmer, and we'll have a look at whether we can do anything about this. I'll let you know what our programmer says.
Ian
RepRapPro tech support