reprappro/RepRapFirmware #81: what's the g-code for color?

columbus1231, 2015-05-19:
I am wondering how color is represented in g-code after conversion from an amf file. 
For example, just like a straight linear movement is represented in g-code as G1 Xnnn Ynnn Znnn Ennn Fnnn Snnn. How is the red green blue (rgb) color code read in g-code by a 3d printer in a section?

RRP-support, 2015-05-19:
This is an automated reply to your message "[RepRapFirmware] what's the g-code for color? (#81)" sent to support@reprappro.com.

Dear columbus1231 <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

RRP-support, 2015-05-19:
Colour is not represented in the gcode file at all... yet. This is mainly because the standard .stl file type does not support colour. All the gcode file can do is change tools, for different stl parts. It is up to you to put the correct colour for the specific part of the print in the right tool (ie, different hot end). Then the gcode changes tool, to use a different extruder/hot end.
If mixing hot ends, ie a single nozzle fed by multiple extruders, become common place, there will have to eventually be some mechanism for this. Our firmware does at least allow you to set up any number of tools, with any mix of extruders (different tools can use different combinations of the same extruder), feeding a single hot end. I'm not sure if you can specify tools with different percentages for extruders (ie rather than 50% extruder 1, 50% extruder 2, you mix 20% and 80%, to get a wider variation of colour), but that is anticipated. You can also manually change the gcode to set different extruder motor speeds, to get different colours, and graduated mixes of colours, but there's no native file format or slicing program that can do this.
In the end, the hardware is there, and it's a software problem. How do you create blended colour in a CAD file, what format do you output to, and how do you slice that object to represent the colour and/or material you want, for the printer firmware to understand?! Tricky problems to resolve!

Ian
RepRapPro tech support

RRP-support, 2015-05-19:
Also, you wouldn't mix Red, green and blue; this is a transmissive colour palette. You would mix White, Cyan, Magenta, Yellow and Black, which is a reflective colour palette (like paper printer print) to get a wide range of colours for 3D objects.

Ian
RepRapPro tech support

columbus1231, 2015-05-20:
Thanks for your help. I will try your suggestion and see on my printer.

RRP-support, 2015-05-20:
This is an automated reply to your message "Re: [RepRapFirmware] what's the g-code for color? (#81)" sent to support@reprappro.com.

Dear columbus1231 <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

closed 2015-05-20 by RRP-support