reprappro/RepRapFirmware #82: File upload sometimes fails

jasontrublu, 2015-05-31:
Hi,

Firmware Version: 	1.09 (2015-04-21)
Web Interface Version: 	HTML: 1.06, JS: 1.06

Sometimes file uploads fail / get stuck. It loads until about 98/99% and then just freezes (without aborting or reporting anything). It seems to happen more often to big files, but bigger files can be successful as well.

RRP-support, 2015-05-31:
This is an automated reply to your message "[RepRapFirmware] File upload sometimes fails (#82)" sent to support@reprappro.com.

Dear Jason Peper <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

chrishamm, 2015-06-03:
How big is the file you were trying to upload? I recently uploaded a file that was 20MB big and it worked just well on my machine. Also, which OS and browser are you using?

RRP-support, 2015-06-03:
This is an automated reply to your message "Re: [RepRapFirmware] File upload sometimes fails (#82)" sent to support@reprappro.com.

Dear Christian Hammacher <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

RRP-support, 2015-06-03:
Hi jasontrublu

We also don't have any problems with big file uploads; v1.09 is much more stable than older versions when it comes to the web interface. A couple of things to check: first, check that you are using the correct SD-Image; if you're using SD files that were meant for an older version of the firmware, they may not work correctly with the current version. ALWAYS use the SD Image supplied with the firmware. If you're using firmware 1.09, the web interface should look like this: https://reprappro.com/wp-content/uploads/2015/04/web-interface-1024x576.png

It could also be a corrupt SD card, particularly if you have had a failed or interrupted file upload. Check the SD card on your PC (use chkdsk) and/or reformat the SD card (copy off any files you have changed first, eg config.g). I find the official SD formatting tool to be the best to reformat the SD cards: https://www.sdcard.org/downloads/formatter_4/index.html

Ian
RepRapPro tech support

jasontrublu, 2015-07-04:
I have not yet have time to investigate this again, as I was on vacation in the UK :)

But the problem is not size alone. I have uploaded bigger files as well. I will write if I had time to investigate further

RRP-support, 2015-07-04:
This is an automated reply to your message "Re: [RepRapFirmware] File upload sometimes fails (#82)" sent to support@reprappro.com.

Dear Jason Peper <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

Moosball, 2015-07-19:
I have the same Problem, im on 1.09 Firmware, using Firefox 39.0. I have formated with the sdcard formatter but no change.
It seems when i upload & print, it works most of the time. but when i just upload, it gets stuck at between 97 and 99 percent. (it was the same gcode file, size 1,3MB).

Regards

RRP-support, 2015-07-19:
This is an automated reply to your message "Re: [RepRapFirmware] File upload sometimes fails (#82)" sent to support@reprappro.com.

Dear Moosball <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

RRP-support, 2015-07-20:
If you've updated to firmware v1.09, have you updated the SD card files as well? Your web interface should look like this: https://reprappro.com/wp-content/uploads/2015/04/web-interface.png 
You MUST use the version firmware and SD card files from the same repository. Do not mix different versions; you may get unexpected results. I say this, because in the current v1.09 web interface, there is no 'upload and print' option, just upload.

Ian
RepRapPro tech support

Moosball, 2015-07-20:
Hi Ian,

Thanks for the Information. My web interface looks excatly like yours in the screenshot. And there is on the Toplevel a Button Upload & Print.

I will try to reinstall the Firmware 1.09, and take the files form the same folder.

In the screenshot you can see the button i was writing about.
[image]

chrishamm, 2015-07-21:
I'm sorry, but we cannot reproduce this issue. That's why I suggest you do the following:

1) Connect to the Duet via USB and leave the serial console open
2) Download Wireshark and start a new session
3) Try to upload a file and if it fails, post the content of your serial log and a copy of your Wireshark dump here

If I discover anything that may be related to the firmware, I'll try to implement a fix and comment on why it didn't work.

RRP-support, 2015-07-21:
Yes, sorry, there is an 'upload and print' button. However, it's function is no different from uploading, and then printing a file, so I don't know why you see different behaviour. It's possible you have a bad sector on the SD card, or that there is a incomplete or corrupt file of the same name that you are trying to over-write. If you can do as chrishamm suggests, we may be able to track down this bug.
Ian
RepRapPro tech support

Moosball, 2015-08-02:
Hello Chrishamm, 

Sorry for the dealy, i hadnt much time the last few days.
As you requested here are some wireshare files. Title of the file says if its working or not, and when it stops / Crash the upload.

The upload source ip is 192.168.1.22 (Client) and Destination is 192.168.1.59 (Printer)

Im not very familiar with wireshare, i caputred all the traffic and uploaded them on dropbox.

https://www.dropbox.com/sh/w2umdgrity32ims/AACA22dGjk-QuH2KgMY7ZjBIa?dl=0

Thanks for your help
Regards,
Steven

chrishamm, 2015-08-02:
Hello Moosball,

No problem, thank you for posting those dumps! I've just looked through them and the problem seems to be that the firmware doesn't send the final JSON response when the upload has finished. The transmission itself looks good to me (I can't tell much about the last 22% dump though), but the question remains why the firmware cannot save/close the file. For this reason I'd like to ask you to perform the following steps:

1) If you have access to a different (class 10) microSD card, please exchange it to exclude the possibility of an SD card failure. Then try to upload a few files once again.

2) When the upload hangs, either open up a new browser window or establish a Telnet connection, send M122 and post its response here. Do NOT press Emergency STOP at this point.

3) Although I'm not convinced this is a browser issue, you could try another one (ie. Firefox instead of Chrome or vice versa). Please check if this makes a difference.

If none of the steps above resolve this problem, I'll prepare another firmware binary for you with some more debug info enabled. This way we can hopefully get a better idea what's going on

dc42, 2015-08-03:
Hi,

Just in case it helps: I had the same problem in a test version of my
fork when I made a change that should have been harmless. See the
comment I added to file Libraries/Lwip/lwip/src/sam/netif/ethernetif.c
in commit
https://github.com/dc42/RepRapFirmware/commit/68cd21c99ceb5f8bd9a5f8f7a419ded528d3f435.
I didn't have time to investigate further.

David Crocker, Escher Technologies Ltd.
http://www.eschertech.com
Tel. +44 (0)20 8144 3265 or +44 (0)7977 211486

RRP-support, 2015-08-03:
This is an automated reply to your message "Re: [RepRapFirmware] File upload sometimes fails (#82)" sent to support@reprappro.com.

Dear dc42 <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

Moosball, 2015-08-30:
Hello Chrishamm, 

Sorry again for the delay, holidays called :)
Well, i had time now and replaced the class 4 Card i got from reprappro. now i have a 16gb class 10 Card insert and made some upload test, one by one, many by one, and it seems to work all the time.
i tryed to Format the class 4 Card, and had no Problem. maybe it has something to do with the transerferspeed through Ethernet.

atm this part works :) lets see if i can find a solution for my calibration Problem :) cant print circles (but i go to the guide befor i ask something new.

have a god time, and thanks to all for the help

regards, 
Steven

Moosball, 2015-09-10:
Hello Chrishamm, 

Well, now i have the same problem again. I dont know what to do. Change my network cable, network switch, and noting works. yesterday i tryed e 13mb file, today its 1,1mb file. I just stops at close to the end of upload.

Maybe you have a chance to build me a firmware with some debug ?

kind regards,
Steven

RRP-support, 2015-09-10:
Copy your 'sys' folder from the SD card to your PC. Reformat the SD card. Copy the SD Image to the card. Copy your old 'sys' folder to the SD card.
Most likely there is a corrupt file that the firmware can't cope with.
Ian
RepRapPro tech support

Moosball, 2015-09-21:
Hi there,
Thanks for you help, but i couldnt reslove it. i get the error on every file that seems to be larger then 1mb. sometimes, after 3 retry it worked, but most i get the error then earlier then 95-99%. then the error comes at 5-8% and i have to poweroff the duet.

any other ideas ?

reagards,
steven

RRP-support, 2015-09-22:
Hi Steven
It really sounds like a physical problem with the SD card or SD socket, and most likely not related to firmware - we just don't see these sorts of problems caused by the firmware. My best guess is that you need to clean the contacts on the SD socket, as there's a poor connection somewhere. 
If you have an aerosol can of Isopropyl Alcohol, disconnect everything from the Duet, and squirt this into the SD card socket, then repeatedly insert and eject the SD card. Wait for the Isopropyl Alcohol to evaporate before reconnecting the Duet. This should clean the contacts.
Also check the soldering of the SD card socket, and check there is nothing bridging the contacts either on the back of the SD card, or pins on the main ARM chip.
Ian
RepRapPro tech support

Moosball, 2015-09-23:
Hello Ian, 
I will give it a try, but i dont think it has something to do with the contacts. Like chrishamm mentioned befor when i send the wireshark log, he sayed he misses the final JSON response.

I also changed the SD Card (Class 10), reformated everything.... and it seemed to be fixed, but then the problem came again.

what i want to do today, maybe i can upload the files over the usb connection from pronterface, and see if i have the same errors, so i can see if i have a problem with my homenetwork.

Regards,
Steven

Chrishamm wrote:
Hello Moosball,

No problem, thank you for posting those dumps! I've just looked through them and the problem seems to be that the firmware doesn't send the final JSON response when the upload has finished. The transmission itself looks good to me (I can't tell much about the last 22% dump though), but the question remains why the firmware cannot save/close the file. For this reason I'd like to ask you to perform the following steps..