reprappro/RepRapFirmware #84: _Calibration-200x200.g Print Fails On Start(Heaters Turn Off)

trybry25, 2015-08-18:
Hi, I'm wondering if anyone can help me with getting my printer to print the 200x200 calibration build. I have already gon through homing X,Y-axises and zeroing Z-axis as,well as bed-plane leveling. I have also gotten my extruder to extrude alittle using the web interface and some G-codes. 
MY PROBLEM IS: when I go to print ._Calibration-200x200.g(or any file) in the web interface, my bed heater and hot-end heater both turn off and nothing happens.
Is there something wrong with the print file? or maybe my config.g? I can't find how to edit the ._Calibration-200x200.g. Can some one tell me where this file is located? or if there exists a newer file i should be using instead?

Thanks, 
Tryg

RRP-support, 2015-08-18:
This is an automated reply to your message "[RepRapFirmware] _Calibration-200x200.g Print Fails On Start(Heaters
 Turn Off) (#84)" sent to support@reprappro.com.

Dear trybry25 <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

RRP-support, 2015-08-18:
Hi Tryg
This isn't a firmware issue, it's a settings problem. Most likely either your hot end 'tool' is not set to T0 that the gcode asks for, or there is some other issue. Please email support@reprappro.com with your problem, what printer you have, and any other relevant setup information. Also please send your config.g file, and tell me what firmware version you are using (send M115 to the printer). Make sure you print the 'Calibration-200x200.g' file, not one named with an '_' at the beginning; these are temporary files, not print files.
Ian
RepRapPro tech support

trybry25, 2015-08-18:
Oh, I seam to have been trying the wrong file. That fixed my problem, although, I'm still having some issues: The calibration peice starts before the hot end is actually hot enough, causing the E-motor to click abit before it actually start extruding. Also, whenever i try to heat bed and hotend simultaniously from room temperature, it overloads my UPS. Is it necessary for the hot end to draw so much power will heating upto standby?
I am using version 1.09 It looks like there is a new version available, I'll check that out and see if it helps

Thanks Again,

RRP-support, 2015-08-18:
This is an automated reply to your message "Re: [RepRapFirmware] _Calibration-200x200.g Print Fails On
 Start(Heaters Turn Off) (#84)" sent to support@reprappro.com.

Dear trybry25 <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

trybry25, 2015-08-18:
and here is my config.g:
; Configuration file for RepRap Mendel 3
; RepRapPro Ltd
;
; Copy this file to config.g if you have a Mendel 3
; If you are updating a config.g that you already have you
; may wish to go through it and this file checking what you
; want to keep from your old file.
; 
; For G-code definitions, see http://reprap.org/wiki/G-code
; 
M111 S0                             ; Debug off
M550 PRepRapPro Mendel 3            ; Machine name (can be anything you like). With DHCP enabled connect to (example) http://reprappromendel3 (machine name with no spaces).
M551 Preprap                        ; Machine password (currently not used)
M540 P0xBE:0xEF:0xDE:0xAD:0xFE:0x14 ; MAC Address
;M552 P0.0.0.0                       ; Un-comment for DHCP
M552  P172.16.138.125   ;P192.168.1.14                  ; IP address, comment for DHCP
M553  P255.255.255.0    ;P255.255.255.0                 ; Netmask
M554  P172.16.138.1     ;P192.168.1.1                   ; Gateway, comment for DHCP
M555 P2                             ; Set output to look like Marlin
G21                                 ; Work in millimetres
G90                                 ; Send absolute corrdinates...
M83                                 ; ...but relative extruder moves
M906 X800 Y1000 Z800 E800           ; Set motor currents (mA)
M305 P0 R4700                       ; Set the heated bed thermistor series resistor to 4K7
M305 P1 R4700                       ; Set the hot end thermistor series resistor to 4K7
M569 P0 S1                          ; Reverse the X motor
M569 P3 S0                          ; Reverse the extruder motor (T0)
M569 P4 S0                          ; Reverse the extruder motor (T1)
M569 P5 S0                          ; Reverse the extruder motor (T2)
M92 E660                            ; Set extruder steps per mm
M558 P2                             ; Use a modulated Z probe
G31 Z1.5 P543   ;Z0.8 P600             ; Set the probe height and threshold (deliberately too high to avoid bed crashes on initial setup)
M556 S75 X0 Y0 Z0                   ; Put your axis compensation here
M201 X500 Y500 Z15 E500             ; Accelerations (mm/s^2)  
M203 X15000 Y15000 Z100 E3600       ; Maximum speeds (mm/min)
M566 X200 Y200 Z30 E20              ; Minimum speeds mm/minute    
M563 P0 D0 H1                       ; Define tool 0
G10 P0 S-273 R-273                  ; Set tool 0 operating and standby temperatures
;M563 P1 D1 H2                       ; Define tool 1, uncomment if you have a dual colour upgrade
;G10 P1 S-273 R-273                  ; Set tool 1 operating and standby temperatures, uncomment if you have a dual colour upgrade
;M563 P2 D2 H3                       ; Define tool 2, uncomment if you have a tri colour upgrade
;G10 P2 S-273 R-273                  ; Set tool 2 operating and standby temperatures, uncomment if you have a dual colour upgrade

trybry25, 2015-08-18:
again, M115: FIRMWARE_NAME:RepRapFirmware FIRMWARE_VERSION:1.09 ELECTRONICS:Duet (+ Extension) DATE:2015-04-21

it looks like the recent update is also version 1.09. What is the difference from what i am using?

trybry25, 2015-08-18:
I think my extrusion problem was with my Z-probe settings, and not the heater.. Looks like this thread can be closed...

Thanks again

closed 2015-08-18 by trybry25

RRP-support, 2015-08-19:
Hi trybry25
Great, glad you got going. The current firmware is v1.09; there may have been an update to the Readme or some other minor file changes to the SD image, but not to the firmware.
It sounds like you were starting the print too low, blocking the nozzle with the bed, but I think you have sorted that out now, too.
If you have any more support-type questions, please email us on support@reprappro.com - the 'Issues' part of github is more for actual firmware-related issues.
Ian
RepRapPro tech support