reprappro/RepRapFirmware #86: M208 fails to set maximum Z movement

gratefulfrog, 2015-11-01:
I am sorry to have to report another issue firmware 1.09,

I would like to set maximum X,Y,Z values to prevent the motors from skipping and/or the head form banging.  However, for the Z axis, the M208 gcode fails to work.

When I run the gcode at the console:
> M208 X131 Y140 Z73

Then I try to manually move the head:
* beyond X=131, the head stops at X=131, and will go no further,
* beyond Y=140, the bed stops at Y=140, and will go no further
* beyond Z=73, **the head continues to climb, with no limit!**

I hope this is not to hard to fix.

Cheers,
Bob

RRP-support, 2015-11-01:
This is an automated reply to your message "[RepRapFirmware] M208 fails to set maximum Z movement (#86)" sent to support@reprappro.com.

Dear gratefulfrog <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

dc42, 2015-11-01:
I just tested it in my fork, and M208 does limit the Z axis movement correctly, assuming of course the printer has been Z-homed and you do not use M564 to disable the limit. However, the current RepRapPro source code explicitly disables the check for the Z axis.

RRP-support, 2015-11-01:
This is an automated reply to your message "Re: [RepRapFirmware] M208 fails to set maximum Z movement (#86)" sent to support@reprappro.com.

Dear dc42 <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

gratefulfrog, 2015-11-01:
Hi dc42,

Of course, I had Z-home before testing...

So you say that your fork corrects this problem as well as the Y compensation issue that I raised earlier?

I'd like to try your fork, could you please tell me where to get it and how to install it, and if there are any issues I should be aware of?

Cheers,
Bob

RRP-support, 2015-11-01:
This is an automated reply to your message "Re: [RepRapFirmware] M208 fails to set maximum Z movement (#86)" sent to support@reprappro.com.

Dear gratefulfrog <notifications@github.com>

Thank you for your email. We endeavour to answer all emails within 24 hours. It may take longer during busy times, and holiday periods. We process all customer queries on a first come first served basis, during our office hours which are 9am to 5pm GMT Mon-Fri.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

dc42, 2015-11-02:
You can find my fork at
https://github.com/dc42/RepRapFirmware/tree/dev/Release. Follow the link
to the version you want (1.09k-dc42 is the latest) then click on Raw to
download it.

If your printer is an Ormerod 2 then your existing config.g file should
work with it. Otherwise, you may need to add a M574 line in config.g to
tell it what sort of homing switches you have. Also if you want to use
simultaneous X and Y homing then you need to use my homing files. See
http://reprap.org/wiki/Configuring_RepRapFirmware_for_a_Cartesian_printer for
more.

David Crocker, Escher Technologies Ltd.
http://www.eschertech.com
Tel. +44 (0)20 8144 3265 or +44 (0)7977 211486