reprappro/RepRapFirmware #89: capability to control the ready to use cutting modules such as auto focus laser head or plasma THC

mkeyno, 2016-03-22:
hi , I was wondering if this firmware have capability to control the ready to use cutting modules such as auto focus laser head or plasma THC , 
I saw couple of effort on Marlin enthusiast to check output sensor the cutting tool and control the stepper motor the module to control height during the printing 
any comments really appreciated

RRP-support, 2016-03-22:
This is an automated reply to your message "[RepRapFirmware] capability to control the ready to use cutting
 modules such as auto focus laser head or plasma THC  (#89)" sent to support@reprappro.com.

Dear mehrdad <notifications@github.com>

This is an automated reply to your message "[RepRapFirmware] capability to control the ready to use cutting
 modules such as auto focus laser head or plasma THC  (#89)" sent to support@reprappro.com.

Dear mehrdad <notifications@github.com>

RepRap Professional Ltd. is now closed.

Note that our closure does not affect RepRapPro China Ltd (http://www.reprappro.com.cn/). That is a completely separate company (though we have obviously worked closely with them) and they will continue.

The market for low-cost 3D printers is now so crowded and so competitive that a small specialist company like ours could not expand. So, because we were not bankrupt and we did not have any debts to pay, we chose to stop while we were ahead and to concentrate on other activities.

We are an open-source company. This means that all our designs, software source-code and documentation are freely available online on our website and on the Github link below. All that information will continue to be available into the future.

Here is a list of places to look if you want to get parts for RepRapPro's designs:

The RepRap.org Website (http://reprap.org/wiki/RepRap_Buyers'_Guide)

eMaker (http://www.emakershop.com/Seller=7244)

RepRap Ltd (http://reprapltd.com/)

RepRapPro China Ltd (http://www.reprappro.com.cn/)

Think3DPrint3D (https://www.think3dprint3d.com/)

Trapletshop (http://gb.trapletshop.com/fisher-delta-3d-printer)

Replikeo (http://www.replikeo.com/en/)

RS Components (http://uk.rs-online.com/web/)

Mouser (http://www.mouser.co.uk/)

Farnell (http://uk.farnell.com/)

Toby (http://www.toby.co.uk/)

Our designs (including those for all the 3D-printed parts) are all available here: Github

If you donâ��t have access to a 3D Printer to make the parts for our designs, try 3Dhubs or your local hackerspace or makerspace. The latter two will also almost certainly have a lasercutter that will make the sheet parts for our designs as well.

We would like to thank everyone who has supported us in the past, either as customers or as contributors who have improved our open-source products, or both. The fact that, unusually for a former company, we will be able to leave megabytes of useful information available to everyone in perpetuity is due in part to the support from those people.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

dc42, 2016-03-22:
RepRapFirmware is now maintained by myself (dc42) and Christian Hammacher (chrishamm), with additional contributions from Dan Newman (who maintains the RADDS port) and Think3DPrint3D. Chris and I maintain separate forks but take most of each other's changes.

I believe the M571 command (http://reprap.org/wiki/G-code#M571:_Set_output_on_extrude) has been used to implement a laser engraver using RepRapFirmware.

RRP-support, 2016-03-22:
This is an automated reply to your message "Re: [RepRapFirmware] capability to control the ready to use cutting
 modules such as auto focus laser head or plasma THC  (#89)" sent to support@reprappro.com.

Dear dc42 <notifications@github.com>

This is an automated reply to your message "Re: [RepRapFirmware] capability to control the ready to use cutting
 modules such as auto focus laser head or plasma THC  (#89)" sent to support@reprappro.com.

Dear dc42 <notifications@github.com>

RepRap Professional Ltd. is now closed.

Note that our closure does not affect RepRapPro China Ltd (http://www.reprappro.com.cn/). That is a completely separate company (though we have obviously worked closely with them) and they will continue.

The market for low-cost 3D printers is now so crowded and so competitive that a small specialist company like ours could not expand. So, because we were not bankrupt and we did not have any debts to pay, we chose to stop while we were ahead and to concentrate on other activities.

We are an open-source company. This means that all our designs, software source-code and documentation are freely available online on our website and on the Github link below. All that information will continue to be available into the future.

Here is a list of places to look if you want to get parts for RepRapPro's designs:

The RepRap.org Website (http://reprap.org/wiki/RepRap_Buyers'_Guide)

eMaker (http://www.emakershop.com/Seller=7244)

RepRap Ltd (http://reprapltd.com/)

RepRapPro China Ltd (http://www.reprappro.com.cn/)

Think3DPrint3D (https://www.think3dprint3d.com/)

Trapletshop (http://gb.trapletshop.com/fisher-delta-3d-printer)

Replikeo (http://www.replikeo.com/en/)

RS Components (http://uk.rs-online.com/web/)

Mouser (http://www.mouser.co.uk/)

Farnell (http://uk.farnell.com/)

Toby (http://www.toby.co.uk/)

Our designs (including those for all the 3D-printed parts) are all available here: Github

If you donâ��t have access to a 3D Printer to make the parts for our designs, try 3Dhubs or your local hackerspace or makerspace. The latter two will also almost certainly have a lasercutter that will make the sheet parts for our designs as well.

We would like to thank everyone who has supported us in the past, either as customers or as contributors who have improved our open-source products, or both. The fact that, unusually for a former company, we will be able to leave megabytes of useful information available to everyone in perpetuity is due in part to the support from those people.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

mkeyno, 2016-03-22:
thanks @dc42 for reply , but unlike the Marlin , I couldn't find the modified RepRap for CNC cutting machine such as laser or plasma , and it seems unlike the 3d Printer  most developer here have own way and not follow the standard same way for CNC cutting machine
actually I made 120 watt  laser CNC with RAMPS and modified Marlin by [ TurnkeyTyranny]( https://github.com/TurnkeyTyranny ) which is equipped with BT wireless link , I used to  repetier host for cutting and engraving. Code has minor defect  and unfortunately not updated any longer , however I'm moving to next step and find many industrial laser CNC have  module which is keep the cutting tool; in readiness state , for example THC module check the correct distance from material constantly , I start sending the note to Marlin and repetier developer to find any enthusiast people to such project and make standard modified version for cutting  CNC machine
I really appreciated to have your comments

RRP-support, 2016-03-22:
This is an automated reply to your message "Re: [RepRapFirmware] capability to control the ready to use cutting
 modules such as auto focus laser head or plasma THC  (#89)" sent to support@reprappro.com.

Dear mehrdad <notifications@github.com>

This is an automated reply to your message "Re: [RepRapFirmware] capability to control the ready to use cutting
 modules such as auto focus laser head or plasma THC  (#89)" sent to support@reprappro.com.

Dear mehrdad <notifications@github.com>

RepRap Professional Ltd. is now closed.

Note that our closure does not affect RepRapPro China Ltd (http://www.reprappro.com.cn/). That is a completely separate company (though we have obviously worked closely with them) and they will continue.

The market for low-cost 3D printers is now so crowded and so competitive that a small specialist company like ours could not expand. So, because we were not bankrupt and we did not have any debts to pay, we chose to stop while we were ahead and to concentrate on other activities.

We are an open-source company. This means that all our designs, software source-code and documentation are freely available online on our website and on the Github link below. All that information will continue to be available into the future.

Here is a list of places to look if you want to get parts for RepRapPro's designs:

The RepRap.org Website (http://reprap.org/wiki/RepRap_Buyers'_Guide)

eMaker (http://www.emakershop.com/Seller=7244)

RepRap Ltd (http://reprapltd.com/)

RepRapPro China Ltd (http://www.reprappro.com.cn/)

Think3DPrint3D (https://www.think3dprint3d.com/)

Trapletshop (http://gb.trapletshop.com/fisher-delta-3d-printer)

Replikeo (http://www.replikeo.com/en/)

RS Components (http://uk.rs-online.com/web/)

Mouser (http://www.mouser.co.uk/)

Farnell (http://uk.farnell.com/)

Toby (http://www.toby.co.uk/)

Our designs (including those for all the 3D-printed parts) are all available here: Github

If you donâ��t have access to a 3D Printer to make the parts for our designs, try 3Dhubs or your local hackerspace or makerspace. The latter two will also almost certainly have a lasercutter that will make the sheet parts for our designs as well.

We would like to thank everyone who has supported us in the past, either as customers or as contributors who have improved our open-source products, or both. The fact that, unusually for a former company, we will be able to leave megabytes of useful information available to everyone in perpetuity is due in part to the support from those people.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

dc42, 2016-03-23:
The person who drove a laser engraver using the M571 code used a standard build of RepRapFirmware. Unlikie Marlin, RRF is designed to allow you to configure all the parameters in a file on the SD card so that users don't need to build the firmware themselves.  Is there something special that you would need in RRF to drive a CNC machine? If you want to pursue this further, please open an issue on the fork of RRF in my github repo.

mkeyno, 2016-03-23:
thanks @dc42  for reply , actually I'v not work with RRF yet , and just look for proper firmware for Due board , I talk with couple of developer here about the idea to make standard firmware for cutting machine as they do for 3d printer , I noticed most modified code has its own approach whereas  industrial CNC cutter software has  same feature , I was hoping talented developer could open thread  here to create standard firmware for cutting machine like laser and plasma , I've already fork the Marlin  for laser machine and I really interested to be help for the RRF approach as well,
my best regards

mkeyno, 2016-03-28:
dear @dc42 can I have your idea about possibility of auto height control of CNC tools in the RRF? 
Right now Manually  Control the laser head height is possible with Marlin function such as [action_laser_focus_custom(),](https://github.com/HakanBastedt/Marlin4Due/blob/Laser/Marlin/ultralcd.cpp#L106-L126)  and I think ,in two ways we can control the  tools heights  during printing our cutting
One with help  of separate modules and one is totally implemented  in firmware
In the first method (hard way) we could use the Arduino module to command the z axis whenever new bias with per set  g-code height come out, and when adjustment performed active the readiness pin  which is controlled by g-code M266 command before each  cutting   
In the second way we could use the function such as [laser_set_focus(currentZ)](https://github.com/HakanBastedt/Marlin4Due/blob/Laser/Marlin/ultralcd.cpp#L1208) in loop function , in this case we need use I2C to request the height from sensor module    

also I really appreciated to  have your colleagues comments

RRP-support, 2016-03-28:
This is an automated reply to your message "Re: [reprappro/RepRapFirmware] capability to control the ready to use
 cutting modules such as auto focus laser head or plasma THC  (#89)" sent to support@reprappro.com.

Dear mehrdad <notifications@github.com>

This is an automated reply to your message "Re: [reprappro/RepRapFirmware] capability to control the ready to use
 cutting modules such as auto focus laser head or plasma THC  (#89)" sent to support@reprappro.com.

Dear mehrdad <notifications@github.com>

RepRap Professional Ltd. is now closed.

Note that our closure does not affect RepRapPro China Ltd (http://www.reprappro.com.cn/). That is a completely separate company (though we have obviously worked closely with them) and they will continue.

The market for low-cost 3D printers is now so crowded and so competitive that a small specialist company like ours could not expand. So, because we were not bankrupt and we did not have any debts to pay, we chose to stop while we were ahead and to concentrate on other activities.

We are an open-source company. This means that all our designs, software source-code and documentation are freely available online on our website and on the Github link below. All that information will continue to be available into the future.

Here is a list of places to look if you want to get parts for RepRapPro's designs:

The RepRap.org Website (http://reprap.org/wiki/RepRap_Buyers'_Guide)

eMaker (http://www.emakershop.com/Seller=7244)

RepRap Ltd (http://reprapltd.com/)

RepRapPro China Ltd (http://www.reprappro.com.cn/)

Think3DPrint3D (https://www.think3dprint3d.com/)

Trapletshop (http://gb.trapletshop.com/fisher-delta-3d-printer)

Replikeo (http://www.replikeo.com/en/)

RS Components (http://uk.rs-online.com/web/)

Mouser (http://www.mouser.co.uk/)

Farnell (http://uk.farnell.com/)

Toby (http://www.toby.co.uk/)

Our designs (including those for all the 3D-printed parts) are all available here: Github

If you donâ��t have access to a 3D Printer to make the parts for our designs, try 3Dhubs or your local hackerspace or makerspace. The latter two will also almost certainly have a lasercutter that will make the sheet parts for our designs as well.

We would like to thank everyone who has supported us in the past, either as customers or as contributors who have improved our open-source products, or both. The fact that, unusually for a former company, we will be able to leave megabytes of useful information available to everyone in perpetuity is due in part to the support from those people.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.