reprappro/RepRapFirmware #90: X endstop always pressed after M569 code

Graeme84, 2016-04-20:
I am working with the build of the software in the `Release` directory and have edited the default `config.g` file that came on the SD card supplied with a Duet 0.8.5 board have bought from Think3dPrint3d.

The default `config.g` file contains configurations for 6 extruders. Even thought I am only using one, I did not see any harm in keeping the other configurations. However, when using the file I found a problem where the software would always behave as though the endstop was pressed. This was not the case when I switched on the board without an SD card present.

Executing the configuration codes one by one, I found the issue began after `M569 P8 S1` (configuring the direction of the 6th extruder). Below is the `pronsole` output of the issue being reproduced, no endstops were pressed during the execution of the codes:

    PC> M119
    SENDING:M119
    Endstops - X: not stopped, Y: at max stop, Z: not stopped
    PC> M569 P8 S1
    SENDING:M569 P8 S1
    PC> M119
    SENDING:M119
    Endstops - X: at min stop, Y: at max stop, Z: not stopped

RRP-support, 2016-04-20:
This is an automated reply to your message "[reprappro/RepRapFirmware] X endstop always pressed after M569 code
 (#90)" sent to support@reprappro.com.

Dear Graeme84 <notifications@github.com>

This is an automated reply to your message "[reprappro/RepRapFirmware] X endstop always pressed after M569 code
 (#90)" sent to support@reprappro.com.

Dear Graeme84 <notifications@github.com>

RepRap Professional Ltd. is now closed.

Note that our closure does not affect RepRapPro China Ltd (http://www.reprappro.com.cn/). That is a completely separate company (though we have obviously worked closely with them) and they will continue.

The market for low-cost 3D printers is now so crowded and so competitive that a small specialist company like ours could not expand. So, because we were not bankrupt and we did not have any debts to pay, we chose to stop while we were ahead and to concentrate on other activities.

We are an open-source company. This means that all our designs, software source-code and documentation are freely available online on our website and on the Github link below. All that information will continue to be available into the future.

Here is a list of places to look if you want to get parts for RepRapPro's designs:

The RepRap.org Website (http://reprap.org/wiki/RepRap_Buyers'_Guide)

eMaker (http://www.emakershop.com/Seller=7244)

RepRap Ltd (http://reprapltd.com/)

RepRapPro China Ltd (http://www.reprappro.com.cn/)

Think3DPrint3D (https://www.think3dprint3d.com/)

Trapletshop (http://gb.trapletshop.com/fisher-delta-3d-printer)

Replikeo (http://www.replikeo.com/en/)

RS Components (http://uk.rs-online.com/web/)

Mouser (http://www.mouser.co.uk/)

Farnell (http://uk.farnell.com/)

Toby (http://www.toby.co.uk/)

Our designs (including those for all the 3D-printed parts) are all available here: Github

If you donâ��t have access to a 3D Printer to make the parts for our designs, try 3Dhubs or your local hackerspace or makerspace. The latter two will also almost certainly have a lasercutter that will make the sheet parts for our designs as well.

We would like to thank everyone who has supported us in the past, either as customers or as contributors who have improved our open-source products, or both. The fact that, unusually for a former company, we will be able to leave megabytes of useful information available to everyone in perpetuity is due in part to the support from those people.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

dc42, 2016-04-20:
Hi Gramme,  the RepRapPro build is very old and predates the Duet 0.8.5 board. I suggest you use my fork or chrishamm's. You can find mine at github.com/dc42/RepRapFirmware. Regards David

RRP-support, 2016-04-20:
This is an automated reply to your message "Re: [reprappro/RepRapFirmware] X endstop always pressed after M569
 code (#90)" sent to support@reprappro.com.

Dear dc42 <notifications@github.com>

This is an automated reply to your message "Re: [reprappro/RepRapFirmware] X endstop always pressed after M569
 code (#90)" sent to support@reprappro.com.

Dear dc42 <notifications@github.com>

RepRap Professional Ltd. is now closed.

Note that our closure does not affect RepRapPro China Ltd (http://www.reprappro.com.cn/). That is a completely separate company (though we have obviously worked closely with them) and they will continue.

The market for low-cost 3D printers is now so crowded and so competitive that a small specialist company like ours could not expand. So, because we were not bankrupt and we did not have any debts to pay, we chose to stop while we were ahead and to concentrate on other activities.

We are an open-source company. This means that all our designs, software source-code and documentation are freely available online on our website and on the Github link below. All that information will continue to be available into the future.

Here is a list of places to look if you want to get parts for RepRapPro's designs:

The RepRap.org Website (http://reprap.org/wiki/RepRap_Buyers'_Guide)

eMaker (http://www.emakershop.com/Seller=7244)

RepRap Ltd (http://reprapltd.com/)

RepRapPro China Ltd (http://www.reprappro.com.cn/)

Think3DPrint3D (https://www.think3dprint3d.com/)

Trapletshop (http://gb.trapletshop.com/fisher-delta-3d-printer)

Replikeo (http://www.replikeo.com/en/)

RS Components (http://uk.rs-online.com/web/)

Mouser (http://www.mouser.co.uk/)

Farnell (http://uk.farnell.com/)

Toby (http://www.toby.co.uk/)

Our designs (including those for all the 3D-printed parts) are all available here: Github

If you donâ��t have access to a 3D Printer to make the parts for our designs, try 3Dhubs or your local hackerspace or makerspace. The latter two will also almost certainly have a lasercutter that will make the sheet parts for our designs as well.

We would like to thank everyone who has supported us in the past, either as customers or as contributors who have improved our open-source products, or both. The fact that, unusually for a former company, we will be able to leave megabytes of useful information available to everyone in perpetuity is due in part to the support from those people.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

Graeme84, 2016-04-20:
Ok, thanks David. I see that there are a lot of forks of this software!

RRP-support, 2016-04-20:
This is an automated reply to your message "Re: [reprappro/RepRapFirmware] X endstop always pressed after M569
 code (#90)" sent to support@reprappro.com.

Dear Graeme84 <notifications@github.com>

This is an automated reply to your message "Re: [reprappro/RepRapFirmware] X endstop always pressed after M569
 code (#90)" sent to support@reprappro.com.

Dear Graeme84 <notifications@github.com>

RepRap Professional Ltd. is now closed.

Note that our closure does not affect RepRapPro China Ltd (http://www.reprappro.com.cn/). That is a completely separate company (though we have obviously worked closely with them) and they will continue.

The market for low-cost 3D printers is now so crowded and so competitive that a small specialist company like ours could not expand. So, because we were not bankrupt and we did not have any debts to pay, we chose to stop while we were ahead and to concentrate on other activities.

We are an open-source company. This means that all our designs, software source-code and documentation are freely available online on our website and on the Github link below. All that information will continue to be available into the future.

Here is a list of places to look if you want to get parts for RepRapPro's designs:

The RepRap.org Website (http://reprap.org/wiki/RepRap_Buyers'_Guide)

eMaker (http://www.emakershop.com/Seller=7244)

RepRap Ltd (http://reprapltd.com/)

RepRapPro China Ltd (http://www.reprappro.com.cn/)

Think3DPrint3D (https://www.think3dprint3d.com/)

Trapletshop (http://gb.trapletshop.com/fisher-delta-3d-printer)

Replikeo (http://www.replikeo.com/en/)

RS Components (http://uk.rs-online.com/web/)

Mouser (http://www.mouser.co.uk/)

Farnell (http://uk.farnell.com/)

Toby (http://www.toby.co.uk/)

Our designs (including those for all the 3D-printed parts) are all available here: Github

If you donâ��t have access to a 3D Printer to make the parts for our designs, try 3Dhubs or your local hackerspace or makerspace. The latter two will also almost certainly have a lasercutter that will make the sheet parts for our designs as well.

We would like to thank everyone who has supported us in the past, either as customers or as contributors who have improved our open-source products, or both. The fact that, unusually for a former company, we will be able to leave megabytes of useful information available to everyone in perpetuity is due in part to the support from those people.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.