reprappro/RepRapFirmware #91: M574 not implemented!

Graeme84, 2016-04-20:
Despite this being used in the example `config.g` bought from Think3dPrint3d, and in various other examples and documentation around the internet, this command is not actually implemented!

Links showing the command in use:
http://reprap.org/wiki/G-code#M574:_Set_endstop_configuration
http://reprap.org/wiki/Configuring_RepRapFirmware_for_a_Cartesian_printer#Movement_section
http://forum.seemecnc.com/viewtopic.php?f=111&t=9299#p79487
http://ormerod9.rssing.com/browser.php?indx=28821977&item=471

RRP-support, 2016-04-20:
This is an automated reply to your message "[reprappro/RepRapFirmware] M574 not implemented! (#91)" sent to support@reprappro.com.

Dear Graeme84 <notifications@github.com>

This is an automated reply to your message "[reprappro/RepRapFirmware] M574 not implemented! (#91)" sent to support@reprappro.com.

Dear Graeme84 <notifications@github.com>

RepRap Professional Ltd. is now closed.

Note that our closure does not affect RepRapPro China Ltd (http://www.reprappro.com.cn/). That is a completely separate company (though we have obviously worked closely with them) and they will continue.

The market for low-cost 3D printers is now so crowded and so competitive that a small specialist company like ours could not expand. So, because we were not bankrupt and we did not have any debts to pay, we chose to stop while we were ahead and to concentrate on other activities.

We are an open-source company. This means that all our designs, software source-code and documentation are freely available online on our website and on the Github link below. All that information will continue to be available into the future.

Here is a list of places to look if you want to get parts for RepRapPro's designs:

The RepRap.org Website (http://reprap.org/wiki/RepRap_Buyers'_Guide)

eMaker (http://www.emakershop.com/Seller=7244)

RepRap Ltd (http://reprapltd.com/)

RepRapPro China Ltd (http://www.reprappro.com.cn/)

Think3DPrint3D (https://www.think3dprint3d.com/)

Trapletshop (http://gb.trapletshop.com/fisher-delta-3d-printer)

Replikeo (http://www.replikeo.com/en/)

RS Components (http://uk.rs-online.com/web/)

Mouser (http://www.mouser.co.uk/)

Farnell (http://uk.farnell.com/)

Toby (http://www.toby.co.uk/)

Our designs (including those for all the 3D-printed parts) are all available here: Github

If you donâ��t have access to a 3D Printer to make the parts for our designs, try 3Dhubs or your local hackerspace or makerspace. The latter two will also almost certainly have a lasercutter that will make the sheet parts for our designs as well.

We would like to thank everyone who has supported us in the past, either as customers or as contributors who have improved our open-source products, or both. The fact that, unusually for a former company, we will be able to leave megabytes of useful information available to everyone in perpetuity is due in part to the support from those people.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

Graeme84, 2016-04-20:
Ok, I see that this is implemented in dc42's fork - https://github.com/dc42/RepRapFirmware/blob/dev/Notes%20on%20dc42%20fork%20of%20RepRapFirmware.pdf

RRP-support, 2016-04-20:
This is an automated reply to your message "Re: [reprappro/RepRapFirmware] M574 not implemented! (#91)" sent to support@reprappro.com.

Dear Graeme84 <notifications@github.com>

This is an automated reply to your message "Re: [reprappro/RepRapFirmware] M574 not implemented! (#91)" sent to support@reprappro.com.

Dear Graeme84 <notifications@github.com>

RepRap Professional Ltd. is now closed.

Note that our closure does not affect RepRapPro China Ltd (http://www.reprappro.com.cn/). That is a completely separate company (though we have obviously worked closely with them) and they will continue.

The market for low-cost 3D printers is now so crowded and so competitive that a small specialist company like ours could not expand. So, because we were not bankrupt and we did not have any debts to pay, we chose to stop while we were ahead and to concentrate on other activities.

We are an open-source company. This means that all our designs, software source-code and documentation are freely available online on our website and on the Github link below. All that information will continue to be available into the future.

Here is a list of places to look if you want to get parts for RepRapPro's designs:

The RepRap.org Website (http://reprap.org/wiki/RepRap_Buyers'_Guide)

eMaker (http://www.emakershop.com/Seller=7244)

RepRap Ltd (http://reprapltd.com/)

RepRapPro China Ltd (http://www.reprappro.com.cn/)

Think3DPrint3D (https://www.think3dprint3d.com/)

Trapletshop (http://gb.trapletshop.com/fisher-delta-3d-printer)

Replikeo (http://www.replikeo.com/en/)

RS Components (http://uk.rs-online.com/web/)

Mouser (http://www.mouser.co.uk/)

Farnell (http://uk.farnell.com/)

Toby (http://www.toby.co.uk/)

Our designs (including those for all the 3D-printed parts) are all available here: Github

If you donâ��t have access to a 3D Printer to make the parts for our designs, try 3Dhubs or your local hackerspace or makerspace. The latter two will also almost certainly have a lasercutter that will make the sheet parts for our designs as well.

We would like to thank everyone who has supported us in the past, either as customers or as contributors who have improved our open-source products, or both. The fact that, unusually for a former company, we will be able to leave megabytes of useful information available to everyone in perpetuity is due in part to the support from those people.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.

dc42, 2016-04-20:
The T3P3 config files are designed for use with my firmware fork.

RRP-support, 2016-04-20:
This is an automated reply to your message "Re: [reprappro/RepRapFirmware] M574 not implemented! (#91)" sent to support@reprappro.com.

Dear dc42 <notifications@github.com>

This is an automated reply to your message "Re: [reprappro/RepRapFirmware] M574 not implemented! (#91)" sent to support@reprappro.com.

Dear dc42 <notifications@github.com>

RepRap Professional Ltd. is now closed.

Note that our closure does not affect RepRapPro China Ltd (http://www.reprappro.com.cn/). That is a completely separate company (though we have obviously worked closely with them) and they will continue.

The market for low-cost 3D printers is now so crowded and so competitive that a small specialist company like ours could not expand. So, because we were not bankrupt and we did not have any debts to pay, we chose to stop while we were ahead and to concentrate on other activities.

We are an open-source company. This means that all our designs, software source-code and documentation are freely available online on our website and on the Github link below. All that information will continue to be available into the future.

Here is a list of places to look if you want to get parts for RepRapPro's designs:

The RepRap.org Website (http://reprap.org/wiki/RepRap_Buyers'_Guide)

eMaker (http://www.emakershop.com/Seller=7244)

RepRap Ltd (http://reprapltd.com/)

RepRapPro China Ltd (http://www.reprappro.com.cn/)

Think3DPrint3D (https://www.think3dprint3d.com/)

Trapletshop (http://gb.trapletshop.com/fisher-delta-3d-printer)

Replikeo (http://www.replikeo.com/en/)

RS Components (http://uk.rs-online.com/web/)

Mouser (http://www.mouser.co.uk/)

Farnell (http://uk.farnell.com/)

Toby (http://www.toby.co.uk/)

Our designs (including those for all the 3D-printed parts) are all available here: Github

If you donâ��t have access to a 3D Printer to make the parts for our designs, try 3Dhubs or your local hackerspace or makerspace. The latter two will also almost certainly have a lasercutter that will make the sheet parts for our designs as well.

We would like to thank everyone who has supported us in the past, either as customers or as contributors who have improved our open-source products, or both. The fact that, unusually for a former company, we will be able to leave megabytes of useful information available to everyone in perpetuity is due in part to the support from those people.

Other support channels are available. Please refer to the relevant troubleshooting guide for your machine:

Ormerod 2: https://reprappro.com/documentation/ormerod-2/troubleshooting/
Ormerod: https://reprappro.com/documentation/ormerod/troubleshooting/

Huxley Duo: https://reprappro.com/documentation/huxley-duo/troubleshooting/
Huxley: https://reprappro.com/documentation/huxley/troubleshooting/

Mono and Tricolour Mendel: https://reprappro.com/documentation/mendel-tricolour/troubleshooting/

Support is also available from other owners of RepRapPro and other 3D printers, either through the RepRap Forums: https://forums.reprap.org or through IRC which can be accessed from here: https://reprappro.com/support-channel/ (works on Chrome, not Firefox).

Best regards,

RepRapPro Support Team.